adafruit/Adafruit_BluefruitLE_nRF51 #2: Characteristic MAX_LEN doesn't seem to take effect

andySigler, 2015-06-06:
Hello, I'm creating my own services and characteristics using the module's AT commands. Below is a printout from when I send "AT+GATTLIST" to the module. As you can see, I've created a service 0x6900, with one "read" characteristic with a value of "hello".

```
ID=01,UUID=0x6900
  ID=01,UUID=0x1110,PROPERTIES=0x02,MIN_LEN=1,MAX_LEN=20,VALUE=hello
```
However, when I connect to the module using the Light Blue app on iOS, the characteristic's value shows up as only the first letter "h". If I change the module's value to "world," then only "w" shows up on the app.

[image]

I do not think this is the app's error, because the UART examples work just fine. I am wondering if it has to do with the MAX_LEN property not being set properly?

The following code is creating the service and characteristic. Before running these lines, I run `.factoryReset()` and send an ""AT+GATTCLEAR". After the following lines, I then run `.reset()`. Both the module's firmware and Adafruit library are up to date, and it's being run on an Arduino Uno.

```
// first, create the service
String serviceString = "AT+GATTADDSERVICE=";
serviceString += "UUID=0x6900";

int serviceID = BLE_print_with_reply(serviceString, 200);
    
// second, create the characteristic
String charString = "AT+GATTADDCHAR=";
charString += "UUID=0x1110";
charString += ", PROPERTIES=0x02"; // read only
charString += ", MIN_LEN=1";
charString += ", MAX_LEN=20";
charString += ", VALUE=hello";

int charID = BLE_print_with_reply(charString, 200);

// below is a helper function to send dynamic strings to the module
// and return the integer value sent back

int BLE_print_with_reply(String msg, int delayTime) {
  delay(delayTime); // optional delay, because it seemed to help communication
  uint32_t reply = -1;
  ble.println(msg);
  ble.sendCommandWithIntReply(F(""), &reply);
  return reply;
}
```

microbuilder, 2015-06-07:
Thanks, I can confirm this is a bug.  When the device resets it uses the min size when populating the characteristic, and doesn't take the max size into account.  As a temporary workaround you'll have to repopulate the value yourself post-reset, but we'll add a fix for this in the next firmware release (which should happen in the next week or so).  I'll close this issue once the firmware update has been released.

andySigler, 2015-06-08:
great thanks!

microbuilder, 2015-07-01:
Updated in the latest 0.6.5 release.

closed 2015-07-01 by microbuilder