adafruit/Adafruit_BluefruitLE_nRF51 #7: peek function fails and stops any more data from being read.

HamAndEggs, 2015-12-15:
If you call peek then it's game over, read will not work.

Here is a use case, I have altered the loop from the bleuart_datamode example. All I done is add a peek() before the read() call and it breaks the app.

void loop(void)
{
  // Check for user input
  char n, inputs[BUFSIZE+1];

  if (Serial.available())
  {
    n = Serial.readBytes(inputs, BUFSIZE);
    inputs[n] = 0;
    // Send characters to Bluefruit
    Serial.print("Sending: ");
    Serial.println(inputs);

    // Send input data to host via Bluefruit
    ble.print(inputs);
  }

  // Echo received data
  while ( ble.available() )
  {
    int x = 'x';
//    x = ble.peek(); // !!!!! Uncomment this line and the app breaks !!!!!
    int c = ble.read();

    Serial.print((char)c);
    Serial.print((char)32);
    Serial.print(x);

    // Hex output too, helps w/debugging!
    Serial.print(" [0x");
    if (c <= 0xF) Serial.print(F("0"));
    Serial.print(c, HEX);
    Serial.print("] ");
  }
}

microbuilder, 2016-02-13:
Can you please confirm that you are using the following:

- latest library 1.7.6
- firmware 0.6.7 (0.7.0 BETA is OK as well).
- Arduino IDE 1.6.5 or 1.6.7

Also, which board are you using (Atmel 32U4 or ATSAMD21 based)?

hathach, 2016-02-13:
Hi,

Could you please post your ATI response (Bluefruit info) as well as the board that you are using. It seems to be OK when I testing with Bluefruit Micro.

HamAndEggs, 2016-02-13:
Hi, this was fixed a while ago. Not sure why the bug was not closed. Was dealt with on the forums. There was a typo in the function and was fixed.

Ta. :)

closed 2016-02-13 by microbuilder