adafruit/Adafruit_BluefruitLE_nRF51 #8: Packet loss and infinite loops

corbinstreehouse, 2015-12-20:
Hi! I noticed several problems when using the library. I will sometimes see the library getting into infinite loops and also loosing packets. It took me a while, but I realized what the problems are (I'll also post on the Adafruit forums soon). I forked and fixed them with minimal changes in https://github.com/corbinstreehouse/Adafruit_BluefruitLE_nRF51, I also have re-write branch that does it better.

Conceptually, here are the issues:
 1. Adafruit_BluefruitLE_SPI::getResponse has this while loop:
   while ( digitalRead(m_irq_pin) &&
          m_rx_fifo.remaining() >= SDEP_MAX_PACKETSIZE )

The problem here is the irq pin might still be left pulled, but data isn't available..this causes problems in getPacket, where it reads the header based on it not reading SPI_IGNORED_BYTE. However, it will eventually get a non-ignored byte from the last packet, and the code:

    while ( p_header->msg_type != SDEP_MSGTYPE_RESPONSE && p_header->msg_type != SDEP_MSGTYPE_ERROR )
    {
      p_header->msg_type = spixfer(0xff);
    }

will skip over bytes loosing data. This is the cause of packet loss.

2. getPacket needs to be rewritten to do the IRQ assertion test at its start. However, I find that on a fast processor (teensy 3.1) it will still be asserted from the last packet. This causes the header to get SPI_OVERREAD_BYTE. If we get SPI_OVERREAD_BYTE , we should wait more on the m_irq_pin. 

Anyways... I'll send you a pull request that fixes this.

hathach, 2016-01-06:
Thank you very much for your effort to investigate on this issue, we have merged you code and do a bit of changes as well. We did change the way library handling the SPI_OVERREAD_BYTE in the https://github.com/adafruit/Adafruit_BluefruitLE_nRF51/commit/4cdb56cb951470144682fb65453ba075b354bea3#diff-a115bc59ca1410371b583f9ab0f72783R578 . It is not a perfect solution but I think a simple delay there should solve the issue (waiting too long will cause timeout in the next loop). The best approach should be using the GPIO interrupt on the IRQ line, but that would require a bit of work since we need to make sure it work with all the boards we produce.

I don't have your Teensy board to test with, if possible, please try to see if that fix your issue or not.

PS: The 0.7.0 beta firmware has a much faster throughput than the current stable one.

corbinstreehouse, 2016-02-17:
Thanks! I did some testing, and I was still seeing packet loss with this version. I need to find out what is different from my branch (which has it solved). I'm now also seeing issues on upload (from the Peripheral to an OS X app). Conceptually, I think the problem is on the SPI side; ```sendPacket``` will start timing out, and the SPI transfer keeps returning SPI_IGNORED_BYTE because I think it is busy. This happens (in my test) when I try to do something like this:
``` C
        char buffer[1024];
        m_ble.write(buffer, 1024);
```
The SPI side just can't seem to handle the 1024 (1kB) write all at once. 

I'll try to isolate and write a test app to show the problem at some point... but if you have any pointers, let me know! I still need to try the beta firmware. maybe that solves my issue. thanks,

corbin

DadoKomar, 2016-03-29:
Hello Corbin.

Thnaks for your update. Also been going through the BLE unit (took me 2 day ;) I had problem, I coudnt sent /n in command mode. So cannot get new line in adafruit Bluefruit LE terminal.
Also it was not sending the lines well, after a time it stops to sending whole line.

So I decided to switch to data mode. It looks better here - more stable. Now testing 1.8.0 version. 
Anyway when I sending over BLE.print() a few chars (30) it takes 30 ms. Which is quite a lot. Dont you know how to speed this up? (It is ok when BLE is still sending, just I would need to execute BLE.print much faster. )

Thanks

corbinstreehouse, 2016-03-29:
Hi Dado,
Yeah --- it has issues! Speed is tricky..you have to do things on both sides to get it to be fast, and you will be limited by the firmware that Adafruit has implemented on the Nordic chip.

Try searching for my blog (corbinstreehouse.com) -- I have some info there and tips on BLE speed.

corbin

DadoKomar, 2016-03-30:
Hello Corbin

Looks it works OK in data mode. I can send special chars also. So if 0.7.0 will improve speed and stability I will stick to this BT.

Cheers,

microbuilder, 2016-09-25:
These issues should be solved in 0.7.0. Issue can be reopened if necessary.

closed 2016-09-25 by microbuilder