adafruit/Adafruit_BluefruitLE_nRF51 #12: SDEP_MAX_PACKETSIZE should be 20.

corbinstreehouse, 2016-02-17:
SDEP_MAX_PACKETSIZE should really be 20. That is because the BLE spec can send up to 20 bytes at a time.  This means that to send a 20 byte packet over BLE with the SPI Friend BlueFruit will take two SPI data packet send/response pairs (since the value is 16):
``` C
#define SDEP_MAX_PACKETSIZE       16 // Maximum payload per packet
```
This is slow and too chatty.

hathach, 2016-02-19:
4 bytes are already taken by SDEP header. Currently we have no time to rework the whole SDEP

closed 2016-02-19 by microbuilder

microbuilder, 2016-02-19:
As Thach mentionned, I think this is unlikely to change in the immediate future, but we can keep it in mind moving forward.  During the design phase there were plans to send SDEP packets over the air as well, which is what was behind the 16+4 byte limit.

corbinstreehouse, 2016-02-20:
Ah, that explains the 20 byte size. Still, it makes it unfortunate for communication over SPI with another microprocessor when you want to send the max BLE packet size (20 bytes).