adafruit/Adafruit_BluefruitLE_nRF51 #17: ble.update() leaks "OK"s

crackmonkey, 2016-07-19:
- Arduino board:  Adafruit Feather 32u4 BlueFruitLE
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.6.8
- Adafruit_BluefruitLE_nRF51 version: 1.9.1
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

I'm using the new callbacks to handle UART RX's and GATT writes, but I am also forwarding traffic between the BlueFruit and Serial so I can talk to the module directly for debugging. (while (ble.available()){Serial.write(ble.read()} and vice-versa). When I first got it running I was getting a constant stream of "OK"s from my reads. I tracked them down to being the responses to println( F("AT+EVENTSTATUS") ); in Adafruit_BLE::update line 284. I think line 285 should be waitForOK(); instead of readline(); so that update() consumes the full result of "AT+EVENTSTATUS". I made the change locally and it seems to work, I can interact with the BLE module without a any excess OK's appearing. It shouldn't require any other changes to the code since waitForOK() puts the data in this->buffer same as readline(). I guess it does cost another BLE_BUFSIZE+1 (49) bytes of ram though.

hathach, 2016-07-19:
Hi,

Thank you for filing the issue. Could you please post your example sketch here. Just to be sure that we are working on the same code.

crackmonkey, 2016-07-19:
[PasteStick - Copy.txt](https://github.com/adafruit/Adafruit_BluefruitLE_nRF51/files/372085/PasteStick.-.Copy.txt)

hathach, 2016-07-20:
thanks, we will check it soon enough :)

hathach, 2016-07-25:
thank you very much for the detail description and the solution. That is an huge bug, nice catch !!! Sorry for getting back late, since we are buried in another yet exciting project :) . I have added waitForOK() at line 285 as you suggested. Thanks again.

commit bfc41133 (2016-07-25): correct callbacks.ino example
fix #17
closed 2016-07-25 by hathach