adafruit/Adafruit_BluefruitLE_nRF51 #19: setBleGattRxCallback fails after multiple calls

microbuilder, 2016-09-20:
See the issue here which was incorrectly filed in the firmware repo: https://github.com/adafruit/Adafruit_BluefruitLE_Firmware/issues/13

collin80, 2016-09-25:
Yeah, sorry, I didn't know the proper repo. I thought the other one would be better just because I think it's a firmware bug not a library issue but I suppose I could be wrong.  Anyone, mostly commenting so I can easily track the status of this potential bug. Thanks for looking into it.

microbuilder, 2016-09-25:
We'll be publishing the Bluefruit firmware shortly with a GPL license (for our own source files), but we are waiting on a legal clearance from Nordic since some of the code we rely on is covered by an older more restrictive license than the last Nordic nRF51 SDKs, and we can't publish as is.  Once the firmware source goes live, that repo will be the right place to post issues, but for now this is probably the best place.

The repo you posted in is really just a place to host firmware, and our mobile apps check there for firmware updates, etc.

Someone will look at this issue this week, though, thanks for filing it.

hathach, 2016-09-28:
hi @collin80 

The issue should be fixed, please
- upgrade the firmware to 0.7.1 and 
- switch the arduino library (this repo) to development branch instead of master

and give it a try again.

microbuilder, 2016-09-28:
You will need to enable BETA firmware updates in the Android or iOS version of Bluefruit LE Connect app to see 0.7.1, just as a FYI!

hathach, 2017-01-19:
@collin80  it has been awhile, however, could you confirm whether the bug is fixed in the latest 0.7.7 ??