adafruit/Adafruit_BluefruitLE_nRF51 #23: flaw in packetParser.cpp 'readPacket' rejects valid packets

ScottFerg56, 2016-12-20:
This file and function appear in three examples that I could find.

The 'readPacket' function resets the packet when a '!' byte is seen:
     if (c == '!') { replyidx = 0; }
    (line 101 after ble->read)
Unfortunately, '!' can be a valid value for a parameter or checksum byte in the packet. This flaw causes certain valid packets to be rejected with a "Checksum mismatch". For example, a color picker packet with any of the RGB values set to 33 ('!') will be rejected. Likewise a color picker packet whose RGB values sum to 122 will generate a checksum value of '!' and be rejected.

petermichaux, 2017-03-16:
I agree this is confusing. It could very well be that a byte contains the value 33 as a color.

petermichaux, 2017-03-16:
@hathach, any insights you can share on this?

ladyada, 2017-03-16:
hathach didnt write it, i did :)