adafruit/Adafruit_BluefruitLE_nRF51 #25: Only the first 7 characteristics are exposed

256dpi, 2017-01-19:
The Adafruit Learn page states the following limits:

```
These values apply to firmware 0.7.0 and higher:
Maximum number of services: 10
Maximum number of characteristics: 30
Maximum buffer size for each characteristic: 32 bytes
Maximum number of CCCDs: 16
```

In my current project I'm registering one service with 9 characteristics, which should be ok. However, only the first 7 characteristics are expose via Bluetooth. The firmware properly assigns a index to the other characteristics and no commands fail, except that they are just not exposed.

Is this somekind of hidden limitation?

Do I need to split the characteristics up in multiple services?

hathach, 2017-01-19:
Could you tell which board you are using and firmware version it is running. If possible please provide your sketch, it is much easier for us to reproduce the issue.

256dpi, 2017-01-19:
Sorry for leaving out some context.

- Host: Arduino MKR1000
- Board: Bluefruit LE SPI Friend
- Firmware: 0.7.7
- Library: 1.9.4

I'm using the `Adafruit_BLEGatt` class to register the services and characteristics.

256dpi, 2017-01-19:
I changed my code to use multiple services. Again all the commands are successful and when I run the `atcommand` sketch and issue `AT+GATTLIST` the following (complete) table is printed:

```
<- ID=01,UUID=0xBA1B,UUID128=63-2F-BA-1B-48-61-4E-4F-81-03-FF-EE-9D-50-33-B5
  ID=01,UUID=0xD327,UUID128=80-2D-D3-27-CA-04-4C-90-BE-86-A3-56-82-75-A5-10,PROPERTIES=0x0A,MIN_LEN=1,MAX_LEN=20,DATATYPE=1,DESCRIPTION=,PRESENTATION=19-00-00-27-00-00-00,VALUE=XXX
  ID=02,UUID=0x3261,UUID128=B3-88-32-61-F3-60-4C-B7-97-91-C3-49-8F-B2-C1-51,PROPERTIES=0x0A,MIN_LEN=1,MAX_LEN=20,DATATYPE=1,DESCRIPTION=,PRESENTATION=19-00-00-27-00-00-00,VALUE=XXX
ID=02,UUID=0x6804,UUID128=6B-0F-68-04-FB-9A-49-C3-B2-F2-85-44-9A-61-B8-11
  ID=03,UUID=0xFFF2,UUID128=19-3F-FF-F2-45-42-4E-BC-BE-1F-4A-35-5D-40-AC-57,PROPERTIES=0x0A,MIN_LEN=1,MAX_LEN=20,DATATYPE=1,DESCRIPTION=,PRESENTATION=19-00-00-27-00-00-00,VALUE=XXX
  ID=04,UUID=0x764C,UUID128=CB-8A-76-4C-54-6C-47-87-9A-6F-42-73-82-D4-97-55,PROPERTIES=0x0A,MIN_LEN=1,MAX_LEN=20,DATATYPE=1,DESCRIPTION=,PRESENTATION=19-00-00-27-00-00-00,VALUE=XXX
  ID=05,UUID=0xE543,UUID128=08-C4-E5-43-65-CC-4B-F6-BB-A8-C8-A5-BF-91-6C-3B,PROPERTIES=0x0A,MIN_LEN=1,MAX_LEN=20,DATATYPE=1,DESCRIPTION=,PRESENTATION=19-00-00-27-00-00-00,VALUE=XXX
  ID=06,UUID=0xA59A,UUID128=AB-B4-A5-9A-85-E9-44-9C-80-F5-72-D8-06-A0-02-57,PROPERTIES=0x0A,MIN_LEN=1,MAX_LEN=20,DATATYPE=1,DESCRIPTION=,PRESENTATION=19-00-00-27-00-00-00,VALUE=XXX
  ID=07,UUID=0xB1B1,UUID128=C5-EC-B1-B1-06-58-4F-C3-80-52-21-55-21-D4-19-25,PROPERTIES=0x0A,MIN_LEN=1,MAX_LEN=20,DATATYPE=1,DESCRIPTION=,PRESENTATION=19-00-00-27-00-00-00,VALUE=XXX
ID=03,UUID=0x1959,UUID128=59-AB-19-59-F7-14-45-C1-A3-39-79-69-A4-B9-29-F6
  ID=08,UUID=0x7850,UUID128=25-42-78-50-28-F3-40-EA-92-6A-AB-3C-EB-6D-0B-56,PROPERTIES=0x0A,MIN_LEN=1,MAX_LEN=20,DATATYPE=1,DESCRIPTION=,PRESENTATION=19-00-00-27-00-00-00,VALUE=XXX
  ID=09,UUID=0x7CE0,UUID128=59-99-7C-E0-3A-50-43-3C-A2-00-D9-F6-D3-12-EB-7C,PROPERTIES=0x12,MIN_LEN=1,MAX_LEN=20,DATATYPE=1,DESCRIPTION=,PRESENTATION=19-00-00-27-00-00-00,VALUE=XXX
OK
```

However, all Bluetooth debug tools I used only show the first two services rather than all three, and the second service is missing the last characteristic. Might this be a firmware issue?

hathach, 2017-01-19:
Can you use Adafruit Bluefruit App to see if it could detect all the service + characteristic. If you could upload your sketch, I am happy to run some tests for you.

256dpi, 2017-02-09:
I spent some more time debuggin this issue and still couldn't find the bug.

The Bluefruit App does not show the missing service and characteristics. I also tested the official "nRF Connect" app for iOS and "LightBlue" for OSX.

[Here](https://github.com/adafruit/Adafruit_BluefruitLE_nRF51/files/763048/BLETest.zip) is the sketch I'm using to test the module. The sketch will perform a factory reset and add 3 services with 2, 5, and again 2 characteristics.

@hathach Thanks for looking into this.

iharshadec, 2022-03-21:
Is this issue fixed in any updated release or still in TODO list?