adafruit/Adafruit_BluefruitLE_nRF51 #27: Make Adafruit_BluefruitLE_SPI an Arduino Stream

monteslu, 2017-02-16:
Enhancement request:

This API is currently already close to Arduino's Stream as is:  https://www.arduino.cc/en/Reference/Stream

If it fully implemented it, we could do fun stuff like plugging it into Firmata and have the uart service pipe data back and forth.

ladyada, 2017-02-16:
we already have a firmata demo, btw

monteslu, 2017-02-16:
Ooh nice!  Want :)

StandardFirmataBLE doesn't support it, so I was trying to get it to work on the Feather 32u4 Bluefruit.

I'm not seeing it anywhere, could you give a URL?

ladyada, 2017-02-16:
https://learn.adafruit.com/bluefruit-le-connect-for-ios/library-and-config

microbuilder, 2017-02-17:
I'm going to close this for now since Firmata exists, but if there is a compelling use case we can open this back up again.

closed 2017-02-17 by microbuilder