adafruit/Adafruit_BluefruitLE_nRF51 #28: adafruit.BLE.cpp troubles

purplefrost, 2017-02-17:
Whenever I try to verify or upload any code from the adafruit bluefruitLE nrf51 I get the error 

C:\Program Files (x86)\Arduino\libraries\Adafruit_BluefruitLE_nRF51\Adafruit_BLE.cpp: In member function 'int Adafruit_BLE::readBLEUart(uint8_t*, int)':
C:\Program Files (x86)\Arduino\libraries\Adafruit_BluefruitLE_nRF51\Adafruit_BLE.cpp:543: error: invalid conversion from 'uint8_t*' to 'char*'
C:\Program Files (x86)\Arduino\libraries\Adafruit_BluefruitLE_nRF51\Adafruit_BLE.cpp:543: error:   initializing argument 1 of 'size_t Stream::readBytes(char*, size_t)'

Is this error from my end or is there a coding error?

ladyada, 2017-02-17:
more information is needed - ide, board, os, etc

microbuilder, 2017-02-17:
Make sure you have the correct target board selected as well, but this is really a support issue. Please file an issue in the Adafruit support forums, and we'll be happy to get this sorted out over there. It's almost certainly a config issue in the IDE. Thanks!

closed 2017-02-17 by microbuilder