adafruit/Adafruit_BluefruitLE_nRF51 #31: What is the motivation for starting GATT values with '!'?

tedhenry100, 2017-03-16:
In the examples, it is not documented why the packet reading code requires all GATT values to begin with the '!' character.

Especially curious is why the `replyidx` is reset when a _second_ '!' character is encountered in the value. What if a '!' character is actually intended to be part of the value the master device is trying to send to the BluefruitLE?

https://github.com/adafruit/Adafruit_BluefruitLE_nRF51/blob/b80b880a62646223ba28b93301164c2134c9f4fc/examples/cplay_neopixel_picker/packetParser.cpp#L101

ladyada, 2017-03-16:
make parsing easier, so you know where to start looking - and then you checksum as well!

closed 2017-03-16 by ladyada

petermichaux, 2017-03-16:
"so you know where to start looking"

If what follows the '!' is the meaningful part of the value sent to the device, is there something in the value that could reasonably come before the '!' character?

petermichaux, 2017-03-16:
I see that issue #23 brings up the same point about resetting when '!' is seen somewhere in the middle of the value. That character might be part of the value rather than the start of the value.

ladyada, 2017-03-16:
that codes probably before the checksum was added. there needs to be *some* way to know when a packet starts! :)

tedhenry100, 2017-03-16:
@ladyada, doesn't data arriving indicate a packet starting?

ladyada, 2017-03-16:
no, not necessarily, bytes can be dropped by the uart interface

tedhenry100, 2017-03-16:
@ladyada, so some of the bytes at the start of the packet could be dropped meaning only the tail of the packet is being read?

That still doesn't explain resetting `replyidx` part way through reading a packet. Can two packets be mashed/merged together requiring the reset? If so, what if the '!' character really is part of the data in the middle of the package.

Thanks for your replies. I appreciate it. :-)

ladyada, 2017-03-16:
yes, if you want to submit a pull req for the fix that would be nice - no ETA on when we'll get to it!