adafruit/Adafruit_BluefruitLE_nRF51 #52: Midi Program Change or 1 byte messages not sent or read

neroroxxx, 2019-09-24:
A midi message received by the BLE module must be at least 3 bytes, if a program change (status+program) is sent by the central the BLE module will ignore it, if a real time like Clock, Start, Continue which are only 1 byte status is sent by the central they are also ignored.

mortenboye, 2020-03-03:
This issue is getting old, but since it is still open I just wanted to note that the best way I found to fix this issue was to reimplement the entire MIDI parsing code myself.

neroroxxx, 2020-03-03:
Where did you do that? I tried just printing out everything on the processRxCallback() in Ble-midi.h but that made no difference, i thought the issue was with directly woth the firmware flashed to the nrf

mortenboye, 2020-03-09:
I did not use any of the MIDI features from this library. Instead I created my own BLE service and characteristic and implemented my own parser according to the BLE MIDI spec.

BassNoize, 2020-04-09:
can you share your code? I'm facing the exact same problem. I want to do the same, implement my own service but if already done... would be nice. Thx

mortenboye, 2020-04-16:
> can you share your code? I'm facing the exact same problem. I want to do the same, implement my own service but if already done... would be nice. Thx

Have a look here https://github.com/InvisibleWrench/BluefruitBLEMIDI/blob/master/example.ino

BassNoize, 2020-04-17:
Dear Morten,

Here is my code:

//For AdaFruit 34u4
#include <Arduino.h>
#include <SPI.h>
#include "Adafruit_BLE.h"
#include "Adafruit_BluefruitLE_SPI.h"
#include "Adafruit_BluefruitLE_UART.h"

#include "BluefruitConfig.h"

#if SOFTWARE_SERIAL_AVAILABLE
#include <SoftwareSerial.h>
#endif

#define FACTORYRESET_ENABLE         1
#define MINIMUM_FIRMWARE_VERSION    "0.7.0"
#define LED     13

Adafruit_BluefruitLE_SPI ble(BLUEFRUIT_SPI_CS, BLUEFRUIT_SPI_IRQ,
BLUEFRUIT_SPI_RST);

bool isConnected = false;
int32_t charid_string;

void setup() {
  pinMode(LED, OUTPUT);

  delay(500);

  Serial1.begin (31250);

  Serial.begin(115200);
  Serial.println(F("Adafruit Bluefruit Own BLE MIDI Service"));
  Serial.println(F("---------------------------------------"));

  /* Initialise the module */
  Serial.print(F("Initialising the Bluefruit LE module: "));

  if ( !ble.begin(VERBOSE_MODE) )
  {
    error(F("Couldn't find Bluefruit, make sure it's in CoMmanD mode &
check wiring?"));
  }
  Serial.println( F("OK!") );

  if ( FACTORYRESET_ENABLE )
  {
    /* Perform a factory reset to make sure everything is in a known state
*/
    Serial.println(F("Performing a factory reset: "));
    if ( ! ble.factoryReset() ) {
      error(F("Couldn't factory reset"));
    }
  }

  if ( !ble.isVersionAtLeast(MINIMUM_FIRMWARE_VERSION) )
  {
    error( F("Callback requires at least 0.7.0") );
  }

  Serial.println( F("Adding Service MIDI with 1 chars props Read | Write
without notify | Notify (0x16)") );
  ble.sendCommandCheckOK(
F("AT+GATTADDSERVICE=uuid128=03-B8-0E-5A-ED-E8-4B-33-A7-51-6C-E3-4E-C4-C7-00")
);
  ble.sendCommandWithIntReply(
F("AT+GATTADDCHAR=UUID128=77-72-E5-DB-38-68-41-12-A1-A9-F2-66-9D-10-6B-F3,
PROPERTIES=0x16, MIN_LEN=1,
MAX_LEN=20,DATATYPE=string,DESCRIPTION=string,VALUE=abc"), &charid_string);
  Serial.println( F("          End Midi") );

  Serial.println();
  ble.sendCommandCheckOK( F("AT+GATTLIST") );
  Serial.println( F("          End Listed Services") );

  ble.sendCommandCheckOK("AT+GAPINTERVALS=8,15,250,180");
  //ble.sendCommandCheckOK(F("AT+GAPINTERVALS=10,2000,200,200"));

ble.sendCommandCheckOK("AT+GAPSETADVDATA=02-01-06-11-06-00-C7-C4-4E-E3-6C-51-A7-33-4B-E8-ED-5A-0E-B8-03");
  ble.sendCommandCheckOK("AT+BLEPOWERLEVEL=0");
  ble.sendCommandCheckOK("AT+GAPDEVNAME=JSB_BLE2MIDI");

  ble.reset();

  /* Disable command echo from Bluefruit */
  ble.echo(false);

  //Serial.println("Requesting Bluefruit info:");
  /* Print Bluefruit information */
  //ble.info();

  /* Set BLE callbacks */
  ble.setConnectCallback(connected);
  ble.setDisconnectCallback(disconnected);
  ble.setBleGattRxCallback(charid_string, BleGattRX);

  ble.verbose(false);

  Serial.print(F("Waiting for a connection..."));
  while (!isConnected) {
    ble.update(500);
  }
}

void loop() {
  ble.update(10);

//  if (! isConnected) {
//    return;
//  }
}

void error(const __FlashStringHelper*err) {
  Serial.println(err);
  while (1);
}

void connected(void) {
  isConnected = true;
  Serial.println(F(" CONNECTED!"));
}

void disconnected(void) {
  Serial.println("disconnected");
  isConnected = false;
}

void BleGattRX(int32_t chars_id, uint8_t data[], uint16_t len)
{
  Serial.print( F("[BLE GATT RX] (" ) );
  Serial.print(chars_id);
  Serial.print(") ");

  if (chars_id == charid_string)
  {
    Serial.write(data, len);
    Serial.println();

    /*    uint8_t * buffer = (uint8_t*)characteristic.value();
                    Serial.print("0x");
                    for( int i = 0; i < characteristic.valueLength(); i++ ){
                        if( buffer[i] < 0x10 ) Serial.print("0");
                        Serial.print( buffer[i], HEX );
                    }
                    Serial.println(); */

    uint8_t * buffer = (uint8_t*)data;
    Serial.print("0x");
    for ( int i = 0; i < len; i++ ) {
      if ( buffer[i] < 0x10 ) Serial.print("0");
      Serial.print( buffer[i], HEX );

      //Copy the character to Serial1 to the Midi out circuit
      Serial1.write (buffer[i]);
    }
    Serial.println();
  }
  /*else if (chars_id == charid_number)
    {
    int32_t val;
    memcpy(&val, data, len);
    Serial.println(val);
    }*/
}

It's not yet well ordered but works! The problem I'm facing thoufg is that
every now and then the service disconnects...

Hope there is something interesting for  you and if any questions...please
feel free

Kind regards, Bas

BassNoize, 2020-04-17:
My service is purely meant for receiving midi program change messages over
bluetooth.
I've got as well a circuit which is translating the 3V3 logic to 5V driven
current source for MIDI; works perfectly
I know midi is supposed to work as well with 3V3 but via this circuit is
more reliable.