adafruit/Adafruit_BluefruitLE_nRF51 #54: Read from Lipo batt level always returns 0

billyjoker, 2020-06-01:
- Arduino board:  Adafruit Bluefruit Feather 32u4 BLE
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.6

Hi, I have tried two ways in order to get a value read from the Lipo:

1) With the official docs, simply as follows:
```
#define VBATPIN A9
float measuredvbat = analogRead(VBATPIN);
measuredvbat *= 2; // we divided by 2, so multiply back
measuredvbat *= 3.3; // Multiply by 3.3V, our reference voltage
measuredvbat /= 1024; // convert to voltage
Serial.print("VBat: " ); Serial.println(measuredvbat);
```
I have read `VBatt: 0`

2) I've executed the code supplied with [battery.ino](https://github.com/adafruit/Adafruit_BluefruitLE_nRF51/blob/master/examples/battery/battery.ino) and I've got an ERROR displayed with the Serial

What could be happening?

billyjoker, 2020-06-10:
I self answer, the way to get the value read is just including the snippet code **inside** the loop of [battery.ino code](https://github.com/adafruit/Adafruit_BluefruitLE_nRF51/blob/d7069a3aa160de74d2b7f32afa78f27d7f5f2891/examples/battery/battery.ino). So, my commented steps must both to work together.

I do not know why the code of [battery.ino](https://github.com/adafruit/Adafruit_BluefruitLE_nRF51/blob/master/examples/battery/battery.ino) is incompleted, so the user must complete the jigsaw reading the [official docs](https://cdn-learn.adafruit.com/downloads/pdf/adafruit-feather-32u4-bluefruit-le.pdf), - page 29 - and to have intuition to get it work properly.

```
#define VBATPIN A9

void loop(void)
{
  // Should get Battery value from LIPO and update
  Serial.print("Update battery level = ");
  Serial.println(value);
  
  battery.update(value);
  
  value--;
  if (value == 0) value = 100;

  float measuredvbat = analogRead(VBATPIN);
  measuredvbat *= 2; // we divided by 2, so multiply back
  measuredvbat *= 3.3; // Multiply by 3.3V, our reference voltage
  measuredvbat /= 1024; // convert to voltage
  Serial.print("VBat: " ); Serial.println(measuredvbat);
  
  delay(5000);
}
```

But it is needed a conversion to a percentage value, and a way to send the level to the device in other thread

isharrrry, 2020-07-10:
What is the reference voltage range used for ADC sampling and the value range of sampling results when using  AT+HWADC command

aidanwhelan, 2023-04-01:
@billyjoker 

If I understand correctly, the two steps you've combined are not meant to do the same thing, and while there may not be any Serial errors, they are not working together:  

From **battery.ino**, the `battery.update()` method only reports what you provide to it; it will change what it reports when you call this method again and make that provided percentage value ready for BLE communication as a battery level. It is up to you to get the actual battery voltage some other way and compute the percentage that goes into `battery.update()`. In your example above, `value` is the variable that should be the calculated percentage of battery capacity, however it currently looks like it just decrements by 1 until it gets to 0, then loops back to 100.  

Separately, the second half of your example (starting from `float measuredvbat = analogRead(VBATPIN);`) is where you actually sample the voltage of your battery. Note, this requires that you use a resistor divider to connect your BAT pin to an I/O pin (in this case, A9). While this `measuredvbat` variable is actually informed by the battery (unlike `value` in the code above), it is not a percentage of the battery capacity, just a voltage. To get a percentage, you will need to estimate using the approximate discharge curve for your battery. For the purposes of calculating battery percentage, it would be amazing if the discharge was linear, however in reality, [LiPoly batteries are 'maxed out' at 4.2V and stick around 3.7V for much of the battery life, then slowly sink down to 3.2V or so before the protection circuitry cuts it off](https://learn.adafruit.com/adafruit-feather-m0-bluefruit-le/power-management).